UbuntuAsahi/ubuntu-asahi #154: Ubuntu 24 server bloqued

krimdev, 2024-11-21:
hi, 
I am on a mac mini m2 2023 
I installed Ubuntu 24 server and when I start it I get this error and it freezes. I can't use the keyboard or see the system in my network.
error : Bluetooth: hc10: Failed to read codec capabilities (-22)
[image]

tobhe, 2024-11-21:
Thanks for the report, this is not something we have seen before. Does it go away if you wait long enough?

krimdev, 2024-11-21:
> Thanks for the report, this is not something we have seen before. Does it go away if you wait long enough?

It's been like this for an hour. Do you think I should wait a little longer?

https://github.com/user-attachments/assets/fd661043-c6bc-47ca-8c38-7ff75c938221

tobhe, 2024-11-21:
as a quick fix you can use ctrl+alt+del to reboot into grub and then press "e" to edit the boot entry, append `bluetooth.blacklist=yes` to the `linux ...` line and ctrl-x to boot. This should get you further while we are trying to find out what's happening there.

edison, 2024-12-25:
@tobhe I have the same issue but keyboard doesn't work in Grub screen. Is there other solution?

I'm about to give up from Ubuntu Server installation,  can't find a way to proceed :(

tobhe, 2024-12-25:
@edison which hardware? mac mini? usb-a or usb-c keyboard? Might make sense to try both and eventually even different orientations for usb-c.

edison, 2024-12-25:
@tobhe it's a Mac Mini M2 A2686. I'm using a Magic Keyboard via USB-A cable.

I don't have a USB-C cable that fits in the keyboard but I'll try other approach. Just gave up of having Ubuntu for the moment.

Maybe I try with some other keyboard. Fun fact is that keyword works with CTRL+ALT+Del at the error screen but then in the Grub it don't.

AraaRashek, 2025-03-12:
Any updates here ?

AraaRashek, 2025-03-12:
> @tobhe I have the same issue but keyboard doesn't work in Grub screen. Is there other solution?
> 
> I'm about to give up from Ubuntu Server installation,  can't find a way to proceed :(

Have you done anything about that one ?

betteray, 2025-03-24:
> [@tobhe](https://github.com/tobhe) it's a Mac Mini M2 A2686. I'm using a Magic Keyboard via USB-A cable.
>
> I don't have a USB-C cable that fits in the keyboard but I'll try other approach. Just gave [...]

same