UbuntuAsahi/ubuntu-asahi #156: Can't update/upgrade

HasanTheSyrian, 2024-12-23:
Updating from apt gives "the repository ... no longer has a release file"

The updater program crashes

Im on Ubuntu 23.10 on an M1 Pro 16"

tobhe, 2024-12-23:
Can you please paste the whole error message? It isn't clear to me which repository exactly is causing the issue.

HasanTheSyrian, 2024-12-24:
Error:
```
Hit:1 http://ppa.launchpad.net/tobhe/asahi/ubuntu mantic InRelease
Ign:2 http://ports.ubuntu.com/ubuntu-ports mantic InRelease
Ign:3 http://ports.ubuntu.com/ubuntu-ports mantic-updates InRelease
Ign:4 http://ports.ubuntu.com/ubuntu-ports mantic-backports InRelease
Ign:5 http://ports.ubuntu.com/ubuntu-ports mantic-security InRelease
Err:6 http://ports.ubuntu.com/ubuntu-ports mantic Release
  404  Not Found [IP: 185.125.190.36 80]
Err:7 http://ports.ubuntu.com/ubuntu-ports mantic-updates Release
  404  Not Found [IP: 185.125.190.36 80]
Err:8 http://ports.ubuntu.com/ubuntu-ports mantic-backports Release
  404  Not Found [IP: 185.125.190.36 80]
Err:9 http://ports.ubuntu.com/ubuntu-ports mantic-security Release
  404  Not Found [IP: 185.125.190.36 80]
Reading package lists... Done
E: The repository 'http://ports.ubuntu.com/ubuntu-ports mantic Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ports.ubuntu.com/ubuntu-ports mantic-updates Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ports.ubuntu.com/ubuntu-ports mantic-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ports.ubuntu.com/ubuntu-ports mantic-security Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

`sources.list`

```
# See http://help.ubuntu.com/community/UpgradeNotes for how to upgrade to
# newer versions of the distribution.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic main restricted

## Major bug fix updates produced after the final release of the
## distribution.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-updates main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-updates main restricted

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team. Also, please note that software in universe WILL NOT receive any
## review or updates from the Ubuntu security team.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic universe
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic universe
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-updates universe
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-updates universe

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team, and may not be under a free licence. Please satisfy yourself as to
## your rights to use the software. Also, please note that software in
## multiverse WILL NOT receive any review or updates from the Ubuntu
## security team.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic multiverse
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-updates multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-updates multiverse

## N.B. software from this repository may not have been tested as
## extensively as that contained in the main release, although it includes
## newer versions of some applications which may provide useful features.
## Also, please note that software in backports WILL NOT receive any review
## or updates from the Ubuntu security team.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-backports main restricted universe multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-backports main restricted universe multiverse

deb http://ports.ubuntu.com/ubuntu-ports/ mantic-security main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-security main restricted
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-security universe
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-security universe
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-security multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-security multiverse
```

tobhe, 2024-12-24:
Makes sense, mantic is out of support. Try upgrading to noble with `do-release-upgrade`

HasanTheSyrian, 2024-12-24:
I already tried it says

```
Checking for a new Ubuntu release
Your Ubuntu release is not supported anymore.
For upgrade information, please visit:
http://www.ubuntu.com/releaseendoflife

Please install all available updates for your release before upgrading.
```

I dont know how Im supposed to update

tobhe, 2024-12-24:
Not yet. It thinks that some package is outdated. Does `apt list --upgradable` give you a hint which package might need updating first?

HasanTheSyrian, 2024-12-24:
It showed me some goofy packages and an ARM package. I did `apt upgrade linux-image-apple-arm` successfully but I still can't update `apt update` and `do-release-upgrade` give the same errors

tobhe, 2024-12-24:
Ok. One more thing you can try: replace all occurrences of `ports.ubuntu.com` in your `sources.list` with `old-releases.ubuntu.com` and see if that allows you to update && upgrade

HasanTheSyrian, 2024-12-24:
Works 🫡

closed 2024-12-24 by HasanTheSyrian