UbuntuAsahi/ubuntu-asahi #160: Getting Kernel panic "Asynchronous SError Interrupt" after upgrading to 6.12.0-1002.2

Rad0van, 2025-01-19:
Getting kernel panic on boot after upgrading to 6.12.0-1002.2. However this happens also if 6.10.0-1001.1 is chosen and even in recovery mode. It is 24.10 KDE desktop upgraded from 24.04. Clueless what to do here.

tobhe, 2025-01-20:
Sorry for that. Can you share the crash log? The more info we have the easier it will be to track down.

theoparis, 2025-01-20:
I encountered the same issue here on a M2 MBP after updating Ubuntu. I do not know how to access the crash logs either, it shows a screen with a ASCII linux penguin at the top left and then says something along the lines of "kernel panic" and "reboot your computer".

jmurudi, 2025-01-20:
I have also encountered same issue now.

tobhe, 2025-01-20:
Working on a fix, I have removed the update from the PPA for now. The fact that selecting an older kernel doesn't work suggests that it might be a device tree problem. I'll let you all know once I know more.

tobhe, 2025-01-20:
@theoparis @jmurudi are your devices also MacBook Pro Max M2?

tobhe, 2025-01-20:
You could try restoring your previous m1n1 from macos for now and see if that helps.

1. Find your EFI partition:
```
diskutil list | grep EFI
   4:                        EFI EFI - UBUNT             500.2 MB   disk0s4
```
2. Mount it (don't copy verbatim, make sure the name is matches the partition found in the previous step)
```
diskutil mount /dev/disk0s4
```
3. Backup m1n1
```
cd /Volumes/EFI\ -\ UBUNT/m1n1
cp boot.bin* ~/
```
4. Restore boot.bin.old
```
mv boot.bin.old boot.bin
```
5. Reboot and see if Ubuntu boots now

Rad0van, 2025-01-20:
> You could try restoring your previous m1n1 from macos for now and see if that helps.
>
> 1. Find your EFI partition:
>
> ```
> diskutil list | grep EFI
>    4:                        EFI EFI - [...]

It does NOT unfortunately :-(

tobhe, 2025-01-20:
@Rad0van Looking at the drm panic documentation it looks like adding `drm.panic_screen=kmsg` to your kernel command line (via grub) might give us some more info.

I have a suspicion what might be causing it. There have been reports of similar errros caused by the controller for the sd-card reader. Try adding `module_blacklist=sdhci_pci` on the command line, that might fix it.

Rad0van, 2025-01-20:
> [@Rad0van](https://github.com/Rad0van) Looking at the drm panic documentation it looks like adding `drm.panic_screen=kmsg` to your kernel command line (via grub) might give us some more info.
>
> I [...]

Blacklisting sdhci_pci does not help. DRM panic screen works, here's what it shows:

[image]

tobhe, 2025-01-20:
So it isn't what I initially suspected. Do you think you can restore `boot.bin` from your macos home directory and see if the error is the same (to make sure we didn't break it by switching to the old one)?

nbosb, 2025-01-20:
I upgraded from Ubuntu 23.10 to 24.04 today and get exactly the same error on a Mac mini M2

[image]

tobhe, 2025-01-20:
Thanks for all the help! I think we found the issue and are working on a fix to test. Issue seems to be related to https://github.com/AsahiLinux/linux/commit/90356e15b4f66f7df2fae4cedb791ebbccebde59

@Rad0van I don't really have an explanation for why the old dtb and kernel don't work either but let's hope the fix still works.

Rad0van, 2025-01-20:
> Thanks for all the help! I think we found the issue and are working on a fix to test. Issue seems to be related to [AsahiLinux/linux@90356e1](https://github.com/AsahiLinux/linux/commit/90356e15b4f66f7df2fae4cedb791ebbccebde59)
> 
> [@Rad0van](https://github.com/Rad0van) I don't really have an explanation for why the old dtb and kernel don't work either but let's hope the fix still works.

Nice. Would be happy to test. But would be curious on how to actually apply the fix...

tobhe, 2025-01-20:
Our fix is an updated m1n1binary with the new device-trees included. You can test it by either manually downloading `https://files.tobhe.de/ubuntu/m1n1.bin` and copying it to your ESP as boot.bin as described above or simply run:
`curl -F https://files.tobhe.de/ubuntu/fix-m1n1 | sh` which should do the same with less typing.

Rad0van, 2025-01-20:
> Our fix is an updated m1n1binary with the new device-trees included. You can test it by either manually downloading `https://files.tobhe.de/ubuntu/m1n1.bin` and copying it to your ESP as boot.bin as described above or simply run: `curl -F https://files.tobhe.de/ubuntu/fix-m1n1 | sh` (from macos) which should do the same with less typing.

Nice. Is the fix already available? Should I try it?

Rad0van, 2025-01-20:
And it WORKS! I needed to apply it manually as my partition is named _EFI - ASAHI_ not _EFI - UBUNTU_ but it works. Thanks a lot!

tobhe, 2025-01-20:
Great :) I'll try to adjust that script and we will make sure to ship an updated kernel later today. Thank you so much for your help debugging this!

john-cabaj, 2025-01-21:
@Rad0van @nbosb @jmurudi @theoparis - just published a new kernel, 6.12.0-1004.4. It should be available for releases 24.04 and newer. Let me know if you have any issues with it, and thanks for the issue reports!

nbosb, 2025-01-21:
i just updated to the new kernel. it seems to work fine on my machine. thanks for the support and the quick fix!

Rad0van, 2025-01-21:
Works fine for me.

murarustefaan, 2025-02-08:
Hey folks,

I tried spinning up a clean installation of Ubuntu 24.04 Server today on a M1 Max Mac Studio and ran across this exact issue. :(
Tries booting, encounters Kernel panic, after a loop of 4-5 reboots with a similar outcome it gets to Recovery Assistant.
[image]

I already tried applying the fix above but to no success. I can provide anything that'll help debugging if you let me know what & how I should do it.

tobhe, 2025-02-08:
@murarustefaan so as I understand it you had no chance applying the kernel update because it crashes right away? I can respin the server build later to make sure it doesn't include a broken kernel in the first place.

murarustefaan, 2025-02-08:
So it crashes right away, but i tried applying it from MacOS using the curl command above couple of comments above which completed successfully. 
However, the outcome is the same, the boot process is interrupted by the kernel crash.

tobhe, 2025-02-09:
@murarustefaan new 24.04 server image is up and should include an up-to-date kernel and m1n1.

murarustefaan, 2025-02-09:
Hey @tobhe, thanks for that. Sadly, it's still not working.

I ran a clean installation, deleting all previous partitions and re-installing 24.04-server from scratch. Now instead of the crash screen I initially posted I get the pink screen [that was posted above](https://github.com/UbuntuAsahi/ubuntu-asahi/issues/160#issuecomment-2602258170).
I also tried re-applying the m1n1 fix above, still to no success.

A couple of reboots into it, it managed to get to this screen, on which it got stuck and had to `crtl+alt+delete` restart it which led back to the pink panic screen.
[image]

Let me know if and how I can provide anything else that could help debugging this issue.

theoparis, 2025-02-25:
I forgot to mention that I received the same error a while ago with a 6.13 Asahi Linux kernel when using Gentoo. Has this issue been raised to the Asahi Linux kernel maintainers?

tobhe, 2025-03-29:
hey all, new server image with latest kernel and a fix fixes is now available. I hope that fixes the remaining issues.

murarustefaan, 2025-03-29:
@tobhe I reinstalled the new image from scratch and it still did not work.

However I moved some stuff around the office and needed to unplug the USB hub I was using and continued the crashing reboot cycle of Ubuntu - asahi as it happened before.
Turns out that was the issue 🤦🏽 , for some reason this kernel does not play well with a USB hub (or with my USB hub, actually).
Now I've a working version of ubuntu on ARM and I could not be happier.

Thanks so much for you work @tobhe!

PS: Another symptom I had was not keyboard input in the EFI Boot screen (where you can toggle between the boot options) which dissappeared once I plugged in the keyboard directly into the Mac.