UbuntuAsahi/ubuntu-asahi #161: Will we have a desktop LTS release at some point?

alessandro-cojp, 2025-01-25:
Hi! I know right now Asahi itself is kind of unfinished and still misses support for things like microphone and Alt DP, however it seems that Hector @marcan is quite close to wrap up the M1 features (except for Touch ID I think) in the near future.

I would love an Ubuntu LTS version of Linux on my Air M1, with 5 years of support, Ubuntu Pro etc...
I know such thing exists for the Server version of Ubuntu with 24.04.1 LTS however, as the title says, would this happen for the desktop version somewhere in the future? What are the main challenges?
Thanks!

tobhe, 2025-01-27:
> Hi! I know right now Asahi itself is kind of unfinished and still misses support for things like microphone and Alt DP, however it seems that Hector @marcan is quite close to wrap up the M1 features (except for Touch ID I think) in the near future.

Indeed things are looking pretty good. In many ways I would say the Asahi experience today is probably already more polished than your average off the shelf random amd64 laptop.

> I would love an Ubuntu LTS version of Linux on my Air M1, with 5 years of support, Ubuntu Pro etc...
I know such thing exists for the Server version of Ubuntu with 24.04.1 LTS however, as the title says, would this happen for the desktop version somewhere in the future? What are the main challenges?

I think you might be mixing up two things here.

There are official Ubuntu server arm64 images at https://cdimage.ubuntu.com/releases/noble/release/. Those are generic arm64 isos, they won't work on Apple Silicon bare metal (they will run fine in a VM though). Starting with 24.10 there is also a generic arm64 desktop iso so 26.04 at latest there will likely also be an arm64 desktop LTS one.

Ubuntu Asahi is a community project providing bare-metal support for Apple Silicon macs. That means we build and maintain packages for all the hardware specific parts Asahi relies on (including their Linux kernel fork) and produce images that "just work". There are even 24.04 images, but there is no official support from Ubuntu/Canonical for those images. You can probably enable Ubuntu pro and get security updates for most packages but not for the hardware specific parts maintained by Ubuntu Asahi out of tree.

In order to get official support for everything we would have to get all those packages into the official Ubuntu archive and find a way to get the images built in the regular pipeline (for example by becoming something like a Ubuntu flavor). We are steadily making progress on this but it is also a huge task, so unfortunately I don't have an ETA.

alessandro-cojp, 2025-01-29:
Thanks for the clarifications! 
So Ubuntu Pro should hypotetically work on the Ubuntu Asahi 24.04 image and provide 5 years of general patches? _How much_ of a bad idea do you think it would be to do that?

My issue is that I'm very used to LTS versions and for my OS to generally "not change" very often. On my x64 desktop I'm still on Ubuntu 20.04 LTS since when that came out and will only update when the standard 5 years of support ends.
Same for the Windows partition, still running LTSC 2019.

I know that Linux on M-series and generally on ARM is a very VERY new thing (I saw your state of Ubuntu speech btw it is quite cool what you're doing), so I also know asking for an LTS whatever now is both out of scope and a probably a bad idea.

My old 2013 Dell Latitude e7240 kicked the bucket and considering that I usually upgrade laptop every ~10 years buying an ARM device like the Air M1 seemed like a good choice, however I didn't realized how much I dislike rolling releases as I'm now on Fedora.

tobhe, 2025-01-30:
>  So Ubuntu Pro should hypotetically work on the Ubuntu Asahi 24.04 image and provide 5 years of general patches? How much of a bad idea do you think it would be to do that?

It's free so not a terrible idea, but keep in mind that this doesn't cover the kernel and boot loaders and it isn't clear and rather unlikely that we (Ubuntu Asahi) will maintain those for the next 5 years.

> My issue is that I'm very used to LTS versions and for my OS to generally "not change" very often. On my x64 desktop I'm still on Ubuntu 20.04 LTS since when that came out and will only upgrade when the standard 5 years of support ends.
Same for the Windows partition, still running LTSC 2019.

Understandable. We have to find a middle way because we are still adding new features and porting things over from the Fedora side. Development usually happens in the latest kernel and mesa and those often need relatively bleeding edge build dependencies (think rust compiler etc.) which again need to be packaged and backported. How long we keep on maintaining the LTS will largely depend on how much work that is. If we have to backport half of 25.04 to keep 24.04 running we might as well ask our users to upgrade to 25.04 instead. Another benefit of using newer releases is that that's usually what the developers use, so you are less likely to hit bugs first. 

> I know that Linux on M-series and generally on ARM is a very VERY new thing (I saw your state of Ubuntu speech btw it is quite cool what you're doing), so I also know asking for an LTS whatever now is both out of scope and probably a bad idea.

Thanks, glad you enjoyed it :)

closed 2025-03-28 by tobhe