UbuntuAsahi/ubuntu-asahi #163: can't install pipx due to python3-venv

chrootchad, 2025-02-02:
```bash
apt-get update && apt-get install pipx
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-venv : Depends: python3 (= 3.12.3-0ubuntu1) but 3.12.3-0ubuntu2 is to be installed
E: Unable to correct problems, you have held broken packages.
```

```bash
Operating System: Ubuntu 24.04.1 LTS              
          Kernel: Linux 6.12.0-1004-asahi-arm
    Architecture: arm64
```

tobhe, 2025-02-03:
Hey, can you share the output of `apt policy`? It seems to work on my 24.04 machine. Is this a new installation?

chrootchad, 2025-02-03:
Thanks for getting back to me. It's a fairly recent install - ubuntu asahi server 24.04 about 5-6 weeks ago.

I actually got pipx working (to the point where I could use it to install what I needed to) by building the latest version of pipx from source following these instructions (but instead using the latest version I could find 1.7.1):
https://github.com/zinc75/pipx-1.6.0-backport-ubuntu-2404-lts?tab=readme-ov-file#building-from-source

Not sure if this is something different from python3-venv but I installed it before i built pipx
```
sudo apt install python3-virtualenv
```

Anyway now pipx is installed and working apt update isn't happy now and wants me to delete it when i run apt --fix-broken install

```
root@ctlocal:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 pipx : Depends: python3-venv but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

Here's the output of apt policy:

```
apt policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 https://ppa.launchpadcontent.net/zhangsongcui3371/fastfetch/ubuntu noble/main arm64 Packages
     release v=24.04,o=LP-PPA-zhangsongcui3371-fastfetch,a=noble,n=noble,l=fastfetch,c=main,b=arm64
     origin ppa.launchpadcontent.net
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/multiverse arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=multiverse,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/universe arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=universe,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/restricted arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=restricted,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/main arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=main,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/multiverse arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=multiverse,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/universe arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=universe,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/restricted arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=restricted,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/main arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=main,b=arm64
     origin ports.ubuntu.com
 510 https://esm.ubuntu.com/infra/ubuntu noble-infra-security/main arm64 Packages
     release v=24.04,o=UbuntuESM,a=noble-infra-security,n=noble,l=UbuntuESM,c=main,b=arm64
     origin esm.ubuntu.com
 510 https://esm.ubuntu.com/apps/ubuntu noble-apps-security/main arm64 Packages
     release v=24.04,o=UbuntuESMApps,a=noble-apps-security,n=noble,l=UbuntuESMApps,c=main,b=arm64
     origin esm.ubuntu.com
1001 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main arm64 Packages
     release v=24.04,o=LP-PPA-ubuntu-asahi-ubuntu-asahi,a=noble,n=noble,l=Ubuntu Asahi,c=main,b=arm64
     origin ppa.launchpadcontent.net
 500 https://download.docker.com/linux/ubuntu noble/stable arm64 Packages
     release o=Docker,a=noble,l=Docker CE,c=stable,b=arm64
     origin download.docker.com
```

tobhe, 2025-02-03:
Weird, so it looks like you are missing the `-updates` pocket for noble which should not cause conflicts but also isn't intentional. Can you check your `/etc/apt/sources.list.d/ubuntu.sources` and make sure the regular ubuntu entry list `noble-updates` under Suites, something like:

```
Types: deb deb-src
URIs: http://ports.ubuntu.com/ubuntu-ports
Suites: noble noble-updates noble-backports
Components: main universe restricted multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg
```

chrootchad, 2025-02-03:
Sorry I forgot I activated an ubuntu pro license on this box.. perhaps that's broken things?

```
cat /etc/apt/sources.list.d/ubuntu.sources

## Ubuntu distribution repository
...
## See the sources.list(5) manual page for further settings.
##
Types: deb
URIs: http://ports.ubuntu.com/ubuntu-ports/
Suites: noble
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

## Ubuntu security updates. Aside from URIs and Suites,
## this should mirror your choices in the previous section.
Types: deb
URIs: http://ports.ubuntu.com/ubuntu-ports/
Suites: noble-security
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg
```

tobhe, 2025-02-03:
I don't think it is your fault, looks like something is wrong in the image build script. Try changing:

```
Suites: noble
```

to

```
Suites: noble noble-updates
```

and then `apt update && apt upgrade`

chrootchad, 2025-02-03:
Ok made the change to /etc/apt/sources.list.d/ubuntu.sources

on apt update && apt upgrade

apt update pulled in a ton of stuff

apt upgrade failed in the same place as expected

```
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 pipx : Depends: python3-venv but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```
Ran apt --fix-broken install which seemed to sort out the python3-venv issue this time instead of wanting to remove pipx

```
~# apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  python3-venv
The following NEW packages will be installed:
  python3-venv
0 upgraded, 1 newly installed, 0 to remove and 124 not upgraded.
1 not fully installed or removed.
Need to get 1034 B of archives.
After this operation, 6144 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ports.ubuntu.com/ubuntu-ports noble-updates/universe arm64 python3-venv arm64 3.12.3-0ubuntu2 [1034 B]
Fetched 1034 B in 1s (1119 B/s)                         
Selecting previously unselected package python3-venv.
(Reading database ... 164241 files and directories currently installed.)
Preparing to unpack .../python3-venv_3.12.3-0ubuntu2_arm64.deb ...
Unpacking python3-venv (3.12.3-0ubuntu2) ...
Setting up python3-venv (3.12.3-0ubuntu2) ...
Setting up pipx (1.7.1-1) ...
needrestart is being skipped since dpkg has failed
```
ran apt upgrade again which proceeded to successfully upgrade a ton of stuff - 
I think this one's sorted, thanks so much for your help!

tobhe, 2025-02-03:
Thanks for helping us with the report, I will see if we need to adjust the image to enable `-updates` by default!

Technically it is an optional feature, security updates come through the `-security pocket` and `-updates` only contains non-security bug fixes, but I don't see why it shouldn't be enabled by default and apparently it is not as optional as it is supposed to be.

chrootchad, 2025-02-04:
>but I don't see why it shouldn't be enabled by default and apparently it is not as optional as it is supposed to be.

Yeah I think it may reduce some silly tickets like this one and improve UX overall - it was a pretty circular issue that lead to me creating this issue, but probably not an uncommon experience.. 
1. I tried to use a generic install bash script to install [ramalama](https://github.com/containers/ramalama) (that failed silently due to pipx not being installed)
2. there was another install option using pip install in their docs... which I tried and I got hit with an 'externally managed environment error' - ref: https://www.reddit.com/r/linux4noobs/comments/1cst9l7/why_when_using_pip_install_on_arch_linux_i_get/
3. that naturally led me back to getting pipx installed and working to substitute the pip install command with (and this ticket eventually getting lodged)

closed 2025-02-06 by chrootchad