UbuntuAsahi/ubuntu-asahi #164: Is normal that the installation takes so long?

MariadlRosa-1224, 2025-02-08:
I'm trying to install ubuntu in a macbook pro m2 (2022) and everything seems to be working except that the installation has been in 61.8% since an hour ago, should I stop and retry? How long does this usually take?

tobhe, 2025-02-08:
@MariadlRosa-1224 Could be that your internet connection to our server is not ideal. We might have to look into finding a CDN solution that works better across the globe. The files are being served from Europe at the moment.

soulteary, 2025-02-16:
I personally recommend downloading all the files locally. Except for the Apple recovery firmware that is relatively large (~10G), the other content is very small. After downloading, start a container service locally, and you can install a device in about 10 minutes.

VSlavskii, 2025-03-14:
It was super slow for me, but after reading this issue,I was able to solve it by connecting to a VPN server in Berlin. As soon as I switched to VPN, the speed went from 150 Kb/s to 15 Mb/s.

tobhe, 2025-03-14:
That would indicate the problem is with the peering between your provider and Hetzner. Adding a VPN just adds an extra hop, it is a bit counter intuitive that that would make it faster.