UbuntuAsahi/ubuntu-asahi #167: Apps freezing

Kruemelmann, 2025-03-19:
Applications freeze

Hello everyone, for a few days now I have had the problem that my applications keep freezing. Especially firefox seems to be affected (but the problem is not limited to the browser). At first I thought it was a browser problem but even after reinstalling it several times it still freezes.

After a little search in the system logs I found the following error:
`
2025-03-19T06:31:47.125974+01:00 localhost tracker-miner-fs-3[149448]: (tracker-extract-3:149448): GLib-
GIO-WARNING **: 06:31:47.125: Error creating IO channel for /proc/self/mountinfo: Invalid argument (g-io
-error-quark, 13) 
`

And after further searching on the internet I came across the issue `https://gitlab.gnome.org/GNOME/localsearch/-/issues/315` which describes the error pretty well. As of the last comment, the problem seems to exist in ubuntu as well, but I only experience it in combination with asahi on the mac (that's why I opened the issue here) but I can move it if you like

tobhe, 2025-03-19:
I think in this case the best way to getting it fixed would be pointing a proper Ubuntu bug on launchpad and linking the upstream discussion. Looks like it was fixed at some point but 24.04 would need a backport.

Kruemelmann, 2025-03-21:
Ok:) to follow up on this one:
I found an older issue in launchpad matching the error (it links back to the issue I added to the description)

I have continued to search for the problem because it makes the browser almost unusable.
Unfortunately, I didn't know exactly where to look for the error, so I checked the system logs.
I noticed an error that is related to wireplumber.

Here is the error:
`
2025-03-21T13:57:20.613459+01:00 localhost wireplumber[2317]: <WpSiAudioAdapter:0xaaaad6eb7c30> tried to link on las
t rescan, not retrying                                                                                              
2025-03-21T13:57:20.613518+01:00 localhost wireplumber[2317]: <WpSiAudioAdapter:0xaaaad6de0db0> tried to link on las
t rescan, not retrying                                                                                              
2025-03-21T13:57:20.613576+01:00 localhost wireplumber[2317]: <WpSiAudioAdapter:0xaaaad6e00f30> tried to link on las
t rescan, not retrying                                                                                              
2025-03-21T13:57:20.613854+01:00 localhost wireplumber[2317]: <WpSiStandardLink:0xaaaad6f17430> 6 of 6 PipeWire link
s failed to activate                                                                                                
2025-03-21T13:57:20.613944+01:00 localhost wireplumber[2317]: <WpSiAudioAdapter:0xaaaad6bd11d0> tried to link on las
t rescan, not retrying   
`

This is only an impression, so especially the first few messages are spammed into the system log.

With that in mind, it's important to note that although I didn't have a microphone before (not really supported by asahi), I did have speaker sound from the built-in system. This no longer works either. In addition, I no longer see a microphone and speaker in the top right of the battery display menu, but only a headset (but I don't have a headset plugged in), so I no longer get any sound.

Could the errors be related because otherwise i don't know where to find the error

I know that the GitHub issues are not a place for individual problems. Honestly I didn't know where to go sorry for that since i cant find the error. I use asahi on a daily basis and hope to find the error somehow because the freezing makes the system almost unusable

Kruemelmann, 2025-03-22:
Thank you very much :) this is very good news because to be honest i was afraid i would have to completely reinstall my system and jump to an old version.

Thank you by the way for your constant effort in this project :) i really enjoy using it.

tobhe, 2025-03-28:
There have been a bunch of package updates this week, I'd expect that those should fix the audio issues you were seeing. Can you confirm that it works after updating?

Kruemelmann, 2025-03-29:
I have updated my system to the latest version. But unfortunately, I still have the problems with the freezing and the sound. I have now narrowed the problem down to the point where I am relatively certain that the apps freeze whenever the application tries to access audio output or input. I took some screenshots after the update, as I am no longer able to change the output via the gui.

[image]

[image]

The last one is a screenshot that captures the behavior from the system logs when an application tries to access an audio device:

[image]

tobhe, 2025-03-29:
Weird. can you check which wireplumber and pipewire version you have installed?

tobhe, 2025-03-29:
a recommendation that has come up in the fedora asahi dev chat a few times with unexplainable sound issues was deleting `~/.local/state/wireplumber` so maybe that helps in your case

Kruemelmann, 2025-03-29:
[image]
[image]

I deleted the folder, but the error still exists. Do you know if it is possible to reset the configs of wireplumber and pipewire to a default state?

tobhe, 2025-03-29:
You could probably delete `~/.config/wireplumber` if it is user configuration or apt purge and reinstall the package. Can you also show `apt list --installed | grep pipewire` and `apt list --installed | grep asahi`?

Kruemelmann, 2025-03-29:
apt list --installed | grep pipewire :
`
╰─$ apt list --installed | grep pipewire

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

gstreamer1.0-pipewire/noble-updates,now 1.0.5-1ubuntu3 arm64 [installed,automatic]
libpipewire-0.3-0t64/noble-updates,now 1.0.5-1ubuntu3 arm64 [installed,automatic]
libpipewire-0.3-common/noble-updates,now 1.0.5-1ubuntu3 all [installed,automatic]
libpipewire-0.3-modules/noble-updates,now 1.0.5-1ubuntu3 arm64 [installed,automatic]
pipewire-bin/noble-updates,now 1.0.5-1ubuntu3 arm64 [installed,automatic]
pipewire-pulse/noble-updates,now 1.0.5-1ubuntu3 arm64 [installed]
pipewire/noble-updates,now 1.0.5-1ubuntu3 arm64 [installed]
`

apt list --installed | grep asahi :
`

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

alsa-ucm-conf-asahi/noble,noble,now 7-1~bpo24.04.1 all [installed,automatic]
asahi-audio/noble,noble,now 3.2-1~bpo24.04.0~asahi0 arm64 [installed,automatic]
asahi-configs/noble,noble,now 1-2 arm64 [installed,automatic]
asahi-scripts/noble,noble,now 20250130-2ubuntu4.24.04 all [installed,automatic]
libegl-mesa0/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
libgbm1/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
libgl1-mesa-dri/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
libglapi-mesa/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
libglx-mesa0/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
linux-asahi-arm-headers-6.12.0-1005/noble,noble,now 6.12.0-1005.5 all [installed,automatic]
linux-asahi-arm-headers-6.13.0-1002/noble,noble,now 6.13.0-1002.2 all [installed,automatic]
linux-asahi-arm/noble,noble,now 6.13.0.1002.2 arm64 [installed,automatic]
linux-headers-6.12.0-1005-asahi-arm/noble,noble,now 6.12.0-1005.5 arm64 [installed,automatic]
linux-headers-6.13.0-1002-asahi-arm/noble,noble,now 6.13.0-1002.2 arm64 [installed,automatic]
linux-headers-asahi-arm/noble,noble,now 6.13.0.1002.2 arm64 [installed,automatic]
linux-image-6.12.0-1005-asahi-arm/noble,noble,now 6.12.0-1005.5 arm64 [installed,automatic]
linux-image-6.13.0-1002-asahi-arm/noble,noble,now 6.13.0-1002.2 arm64 [installed,automatic]
linux-image-asahi-arm/noble,noble,now 6.13.0.1002.2 arm64 [installed,automatic]
linux-modules-6.12.0-1005-asahi-arm/noble,noble,now 6.12.0-1005.5 arm64 [installed,automatic]
linux-modules-6.13.0-1002-asahi-arm/noble,noble,now 6.13.0-1002.2 arm64 [installed,automatic]
m1n1/noble,noble,now 1.4.21-1ubuntu0.1~asahi1 arm64 [installed,automatic]
mesa-libgallium/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
mesa-va-drivers/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
mesa-vdpau-drivers/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
mesa-vulkan-drivers/noble,noble,now 25.0.0~pre20241211-1asahi0.24.04 arm64 [installed,automatic]
triforce-lv2/noble,noble,now 0.2.0-1~asahi0.24.04 arm64 [installed,automatic]
u-boot-asahi/noble,noble,now 2024.04-1ubuntu0.1~asahi2 arm64 [installed,automatic]
ubuntu-asahi/noble,noble,now 1.26 arm64 [installed]
`

I have tried to delete and reboot, but problem still exists:/

tobhe, 2025-03-29:
hm I think I see what the problem might be. Do you have a `/etc/apt/preferences.d/ubuntu-asahi.pref` file on your system? that SHOULD be part of the ubuntu-asahi package.
Your pipwire version is not the one I'd expect. It should be `pipewire - 1.0.5-1ubuntu2.0asahi0` if the pinning works correctly...
easy to fix with an update in the ppa but I wonder why the pinning didn't work in the first place.

Kruemelmann, 2025-03-29:
I dont have a '/etc/apt/preferences.d/ubuntu-asahi.pref' but after some digging in the system i found out that i have a '/etc/apt/preferences.d.save/preferences.d/ubuntu-asahi.pref'
the content of this is:
`
Package: *
Pin: release o=LP-PPA-ubuntu-asahi-ubuntu-asahi
Pin-Priority: 1001
`

do you know why this happens? can i simply move the *.save folder so that the path matches the one you wrote to me?

tobhe, 2025-03-29:
> do you know why this happens? can i simply move the *.save folder so that the path matches the one you wrote to me?

yes, that should work

It might be a problem related to release upgrades, I'll have to dig a little deeper to see why it might have moved that. We specifically configure the release updater not to touch the ppa config.

Kruemelmann, 2025-03-29:
Ah just found out that it could be a path problem, maybe. 

I have a /etc/apt/preferences.d/preferences.d/ubuntu-asahi.pref

I will try to move it and see if it works after that.

tobhe, 2025-03-29:
huh yes that is wrong

Kruemelmann, 2025-03-29:
Ok yes that fixed the sound problem:) the output sound works fine. I think I'll wait for a few weeks to see if the upcoming changes in asahi-audio fix the mic.

I've already said it, but I'd like to say it again: Thank you for your patience with me.

I assume that the freezing should now also be solved. So I'll close the issue and if it still happens I'll just open it again :)
As I said, thank you very much:)

closed 2025-03-29 by Kruemelmann

tobhe, 2025-03-29:
Well thanks you for helping me with all the info! I'll try to find the root cause and see if we can fix this properly.

tobhe, 2025-03-29:
Oh one more thing you could maybe check for me. What does `dpkg -L ubuntu-asahi` print? wonder if that also shows the `preferences.d/preferences/d`.

Kruemelmann, 2025-03-29:
This looks fine i think:

[image]

tobhe, 2025-03-29:
Funny. so that shows the correct file path. Thanks, I think I have everything I need now!