UbuntuAsahi/ubuntu-asahi #169: Touchpad and wifi not working in 6.13.0-1002 for 13" M2 MBA

MasterDuke17, 2025-03-29:
I have  13" M2 Air, and when I rebooted into the new 6.13.0-1002 kernel my touchpad and wifi weren't working. I looked at dmesg and saw some what I believe are new errors about hid devices not loading, but unfortunately I didn't save the exact output. KDE's system setting didn't even think there was a touchpad at all. Everything worked fine when I re-rebooted back into an older kernel.

MasterDuke17, 2025-03-29:
Here are my most recent couple boot logs (`journalctl -b` through `journalctl -b -5`).
[boot0.log](https://github.com/user-attachments/files/19514360/boot0.log)
[boot1.log](https://github.com/user-attachments/files/19514361/boot1.log)
[boot2.log](https://github.com/user-attachments/files/19514362/boot2.log)
[boot3.log](https://github.com/user-attachments/files/19514365/boot3.log)
[boot4.log](https://github.com/user-attachments/files/19514364/boot4.log)
[boot5.log](https://github.com/user-attachments/files/19514363/boot5.log)

tobhe, 2025-03-29:
It looks like the firmware loading failed, which would explain the symptoms.
A few more things that would help:
- check if find /lib/firmware/vendor/ and mount | grep vendorfw actually shows the firmware directory
- if you add debug to the kernel command line you should get a initramfs log in /run/initramfs/initramfs.debug, that might help us figure where it is failing
- which version of asahi-scripts is installed?

MasterDuke17, 2025-03-29:
It turns out asahi-scripts and ubuntu-asahi had been removed during a previous `apt full-upgrade`. After I reinstalled those, everything is working fine in the new kernel.

closed 2025-03-29 by MasterDuke17