UbuntuAsahi/ubuntu-asahi #171: Fix instructions

suoko, 2025-04-11:
Below I report the instructions you write before the first reboot but I had a slightly different xp, I didn't have to choose a volume to recover.
Point 6 in my case was like this: 
- The mac entered in recovery mode
- I had to choose one MacOS account (I had more then once) and put its credentials
- After that a terminal has automatically started and the Asahi script run in it

_When the system shuts down, follow these steps:
...
6. You will briefly see ,a "macoS Recoveryi dialog.
* If you are asked to."Select a volume to recover!
then choose your normal macos volume and click Next.
You may need to authenticate yourself with your macos credentials.
..._

tobhe, 2025-04-11:
This is unexpected. normally there shouldn't be a recovery dialogue. Maybe something went wrong somewhere and you had to restart in between? I am not 100% sure how macos determines when to show this but it has some kind of automated crash detection.

tobhe, 2025-07-20:
Closing because the issue seems to be resolved

closed 2025-07-20 by tobhe