UbuntuAsahi/ubuntu-asahi #172: Internal HID is broken after upgrading to 25.04

amamegs, 2025-05-02:
I use ubuntu asahi with the kubuntu desktop environment, I have just upgraded to plucky puffin (at least a proposed release) and the update went fine, but when I got to the log in page I was suddenly unable to input my password, so I tried connecting an external keyboard which failed but after restarting (the power button thankfully still works) I got to grub, where I can input using the external keyboard (as long as it was connected before u-boot loaded) but didn't really lead me anywhere, I did get to the recovery mode but I didn't really know what to do.

tobhe, 2025-05-02:
What would help is your upgrade log from `/var/log/dist-upgrade` and maybe the output of `apt list --installed` to see which packages are still there.

There have been problems with kubuntu which is why auto upgrades are still disabled, see https://discourse.ubuntu.com/t/status-of-oracular-to-plucky-upgrades/59652.
Since you have been running kde I wonder if this might be a variant of https://bugs.launchpad.net/ubuntu/+source/ubuntu-release-upgrader/+bug/2106202. If yes you could probably reinstall kubuntu-desktop.

tobhe, 2025-05-04:
I think the bug is the same as on kde but it manifests differently in your case.

> 2025-05-02 14:05:25,923 DEBUG Obsolete: 2to3 asahi-configs fex-emu-armv8.4 fex-emu-binfmt32 fex-emu-binfmt64 idle-python3.12 imagemagick-6.q16 kerneloops kf5-kirigami-addons-data latte-dock legcord libapt-pkg6.0t64 libassuan0 libconfig++9v5 libddcutil4 libflac12t64 libfmt9 libfwupd2 libgbinder libgdal35 libgeos3.12.2 libgit2-1.7 libglibutil libgspell-1-2 libhdf5-103-1t64 libhdf5-hl-100t64 libicu74 libimobiledevice6 libjsoncpp25 libjxl0.10 libkaccounts2 libkf5kdegames-data libkf5kdegames7 libkf5sane-data liblept5 libmagickcore-6.q16-7-extra libmagickcore-6.q16-7t64 libmagickwand-6.q16-7t64 libmbedcrypto7t64 libmsgraph-0-1 libmysqlclient21 libnetcdf19t64 libokular6core1 libopencv-calib3d406t64 libopencv-contrib406t64 libopencv-core406t64 libopencv-dnn406t64 libopencv-features2d406t64 libopencv-flann406t64 libopencv-highgui406t64 libopencv-imgcodecs406t64 libopencv-imgproc406t64 libopencv-ml406t64 libopencv-objdetect406t64 libopencv-video406t64 libopencv-videoio406t64 libpaper1 libperl5.38t64 libpoppler-cpp1 libpoppler140 libpython3.12-dev libpython3.12-minimal libpython3.12-stdlib libpython3.12-testsuite libpython3.12t64 libqpdf29t64 libqt5webkit5 libquotient0.7 librhash0 libroc0.3 libsgutils2-1.46-2 libspdlog1.12 libsubid4 libsuperlu6 libtag1v5 libtag1v5-vanilla libusbmuxd6 libx265-209 libzip4t64 linux-asahi-arm linux-asahi-arm-headers-6.12.0-1005 linux-asahi-arm-headers-6.13.0-1002 linux-headers-6.12.0-1005-asahi-arm linux-headers-6.13.0-1002-asahi-arm linux-headers-asahi-arm linux-image-6.12.0-1005-asahi-arm linux-image-6.13.0-1002-asahi-arm linux-image-asahi-arm linux-modules-6.12.0-1005-asahi-arm linux-modules-6.13.0-1002-asahi-arm perl-modules-5.38 policykit-1 python3-lib2to3 python3.12 python3.12-dev python3.12-doc python3.12-examples python3.12-full python3.12-gdbm python3.12-minimal python3.12-tk python3.12-venv qml-module-org-kde-kaccounts qml-module-org-kde-kirigami-addons-components qml-module-org-kde-kirigami-addons-delegates qml-module-org-kde-kirigami-addons-labs-components qml-module-org-kde-kirigami-addons-labs-mobileform qml-module-org-kde-kquickimageeditor

There are a few things in here that totally shouldn't have been removed. You can probably fix it by going into the recovery again, mounting the original disk and then chrooting to your installed system and installing the missing dependencies.

Basically something like:
```
# mount /dev/nvmen0pX /mnt
# mount /dev/nvmen0pX+1 /mnt/boot
# mount /dev/nvmen0pX+2 /mnt/boot/efi

# mount --bind /dev /mnt/dev
# mount --bind /proc /mnt/proc
# mount --bind /sys /mnt/sys

# chroot /mnt
# apt install ubuntu-asahi linux-apple-arm
```

amamegs, 2025-05-06:
Sorry again for the late response.

Managed to get it to work by enabling network, running dpkg and removing the .save extension from the ubuntu asahi repo, it installed and I used apt upgrade but it didn't do anything and I don't know how to show the output of apt list --installed without having access

tobhe, 2025-05-06:
huh how did that happen. There should be a config in `/etc/update-manager/release-upgrades.d/` that ensures the ppa does not get disabled on upgrades. I think it is shipped by the `ubuntu-asahi` meta package. Maybe the package was somehow removed before you upgraded.

amamegs, 2025-05-06:
I think it got re-added when I installed thw packages (I pressed y to install the original files)

tobhe, 2025-05-06:
Well I'm glad you managed to solve the mystery! Hope that this isn't too common. At least we now have a documented solution here thanks to you.

amamegs, 2025-05-07:
Actually just in case anyone reads this and it ends up unclear here's a set by step guide:
1. Open recovery mode (if you are able to log in normaly then just skip to step 3) - hold esc while booting then in grub (you might need an external keyboard) open the extra options and then the latest version labled recovery mode
2. Connect your mac to an ethernet cable and then choose the network option and then the dpkg option (if dpkg shows packages that need to be upgraded you should probably do that)
3. Open up a shell via the root button on the recovery mode screen and use (sudo) mv /etc/apt/sources.list.d/ubuntu-asahi.list.save /etc/apt/sources.list.d/ubuntu-asahi.list
4. Then finally, (sudo) apt install --reinstall ubuntu-asahi

Anyway do you have any idea as to why I still don't have hid working?

amamegs, 2025-05-19:
Just a quick update to show what I have tried between the last message and now.

1. I have removed i386 and amd64 packages (which I wanted to use for steam and the system gave me a hard time uninstalling)
2. I tried to manually install all of the qt dependencies
3. finally noticed the errors on boot: these happen after grub and the kernel I choose (and if I choose recovery mode or normal mode) "[ERROR] could not allocate <number of bytes in hex, different for every boot> bytes to 0xffffffff" this repeats at least 10 times during boot. there is also "EFI_RNG_PROTOCOL not found" which shows up once in the boot process

I still don't have a working linux install😞

amamegs, 2025-05-22:
U-boot 2025.01-1~0ubuntu (Feb 27 2025 - 17:00:21 +0000)

tobhe, 2025-05-23:
interesting, that is unexpected. can you run `apt policy` and show the result? also `apt-cache madison u-boot-asahi`

amamegs, 2025-05-23:
Ok, I think we found it. U-boot is installed twice. I have attached pictures of the outputs (sorry for no quality). P.s. I wanted to remove the old U-boot to test that but I couldn't figure it out without removing both U-boot's and ubuntu-asahi

[image]
[image]

tobhe, 2025-05-26:
What pulls in regular` u-boo`t? `u-boot-asah`i is the one I would expect to be there

amamegs, 2025-05-27:
Sorry, I didn't undrrstand what to do

tobhe, 2025-06-02:
can you paste the output of `apt list --installed | grep u-boot`?

amamegs, 2025-07-03:
Do I give up now or do you think this is solvable? I can still only interface u-boot with an external keyboard

amamegs, 2025-07-03:
Ok, I have reinstalled every package in the ubuntu asahi launchpad that was installed in two versions or in the wrong version, and now I can interface u-boot without an external keyboard. Once booted nothing changed, just the login screen waiting for input that I cannot give.

amamegs, 2025-08-06:
In case anyone else is experiencing this issue:
After everything in here I decided to just try running on gnome to see what happens and I got in. In fact the issue was entirely within sddm and even when using kde with gtk3 (that might not be the right name so sorry) it ran normaly. This let me backup my important files via gui and reinstall so I can get sddm back.

closed 2025-08-06 by amamegs