UbuntuAsahi/ubuntu-asahi #173: microphone and built-in camera app not working

nonresidentaliennico, 2025-05-19:
hey! 

i installed ubuntu 24.10 on my machine (macbook pro m2 max). everything works perfectly except for the built-in camera app (the camera does work on browser but sometimes it's not recognized) and the microphone does not work at all. the system simply does not pick it up. 

is there a workaround for this? 

thanks a lot in advande!

tobhe, 2025-05-20:
Thx for the feedback! The gnome camera app seems to have a bug where it doesn't work properly with the camera on those machines yet. For the microphone if I remember correctly there was a known issue on the M2 Pro max too, afaik a fix in the form of a kernel update is WIP.

reviczky, 2026-01-19:
Fixed PipeWire version 1.5.85 (https://gitlab.freedesktop.org/pipewire/pipewire/-/releases/1.5.85) is now also packaged for Debian: https://tracker.debian.org/news/1708643/accepted-pipewire-1585-1-source-into-experimental/