UbuntuAsahi/ubuntu-asahi #181: Installing vlc crashes X configuration

mpgalaxy, 2025-09-26:
Hi,

I installed Ubuntu desktop 25.04 on my M1, worked flawlessly.
Running ubuntu I did a (in terminal)

```
apt-get update && apt-get upgrade
```
followed by

```
apt-get clean && apt-get autoremove
```
Even after reboot everything is working perfectly.
Hardware acceleration for youtube is working perfectly fine, although DRM in firefox and chromium is not working and I couldn't get it to work.

Then I installed vlc via apt and on next reboot the X configuration was completely broken due to driver issues, I was stuck in console.

Could you please investigate / check why this is happening exactly ?

tobhe, 2025-09-26:
Thanks, I'll take a look. Can you extract some logs from your system, in particular `/var/log/apt`?

mkurz, 2025-09-26:
> Hardware acceleration for youtube is working perfectly

Hard to believe since the drivers do not exist yet.

Are you not using Wayland?

mkurz, 2025-09-26:
> I'm using the ubuntu image as provided. Playing 4k streams in youtube on full screen the cpu cores are between 20 and 30 %, mostly at 20/25 (using firefox).

That does not mean acceleration acceleration is enabled. The Video decoder is still WIP: https://asahilinux.org/docs/platform/feature-support/m1/#table-of-contents
You are using the CPU to decode video, not the gpu.

mpgalaxy, 2025-09-28:
> Interesting, yes it does indeed look like a mesa issue. I wonder if installing mesa from this ppa that always backports the latest version helps: https://launchpad.net/~kisak/+archive/ubuntu/kisak-mesa

Hi, yes, maybe this could help. As I reinstalled ubuntu-sashi about 10 times in the last 4 days I'll stay with the current installation until there is a fix available. This really decimates the lifetime of the soldered in thus not exchangeable ssd of the mac and booting MacOS from an external SSD is not supported.

Looking forward to your results...