UbuntuAsahi/ubuntu-asahi #182: Is Ubuntu Asahi dead? What's the support status of 25.04?

Rad0van, 2025-09-30:
Hi,

wanted to ask what is the state of Ubuntu Asahi. There has been no commit since February. Last supported version is 24.10 which is no longer receiving updates from Ubuntu. There are some issues reported for 25.04 so not sure if it's safe to install.

Don't get me wrong: I am not complaining. I am extremely grateful for the effort but would like to know if it continues or not and I should be looking for alternatives. Thank you.

tobhe, 2025-09-30:
Not dead, but working on 25.10 which comes out in a few weeks. The repo itself only gets updated when we spin new images which isn't generally needed very often. The old ones should still work and after a package update get all the latest stuff too.

24.10 being the latest supported version is just a bug in the readme, the installer should offer 25.04.

Rad0van, 2025-10-01:
Hi,

thanks for quick reply. So even upgrade to 25.04 should work fine? Want to be sure before I start it. Thanks again.

tobhe, 2025-10-01:
I have not encountered anything like it on my machine but I also don't use bluetooth a lot except for audio where we have a known issue that makes it crackle when used at the same time as wifi. That is a driver problem though.

tobhe, 2025-12-07:
> Hi @tobhe . As the system keeps nagging me that support for 25.04 ends in few days I'd like to ask what is the status of 25.10? It is officially out for almost 2 months (again as in my original post: not complaining, being thankful). Is it safe to upgrade? I am very cautious about this now as my old Lenovo T450 gave out and the only machine I have now is MacBook Pro M2 Max 96GB RAM running Ubuntu Asahi with KDE desktop. So cannot afford to be left with nonworking machine.

I've been running 25.10 since a while now, upgraded shortly before the actual release to make sure everything works. afaict there shouldn't be any issues.

> I know it doesn't belong here but I was trying (with no luck) to install widewine from your repo https://github.com/tobhe/widevine-installer The issues are not enabled there. Is there a way to report it not working or get help somewhere?

One of the problems there was ABI incompatibilities between the snap baseline and widevine. In the past updating chromium/firefox to beta channels fixed some of the issues.

stephan1827, 2025-12-09:
I am using a macmini M1 and installed 25.04 and upgraded to 25.10, but then got weired graphic issues where the mouse pointer would create a trailof arrows. I am not sure if that is related to 25.04 or 25.10 since I did not test 25.04 very long. During the update I received a warning that some packages that referenced Asahi may not be compatible. Should we hold off on 25.10 for now or should this work?